openhwfoundation/cv32e40p #1041: Error during Elaboration in Synopsys DC (ELAB-384)

omarsalaheldinnn, 2025-04-18:
Error: /.../srcs/cv32e40p/cv32e40p_fp_wrapper.sv:91: Too few elements in assignment pattern expression list (ELAB-384) [at this line](https://github.com/openhwgroup/cv32e40p/blob/master/rtl/cv32e40p_fp_wrapper.sv#L91)

I am wondering if this was intended to be written like this: `PipeConfig: '{ default: fpnew_pkg::AFTER }`

Or, should I explicitly list each element in a specific order to match the number of elements in the array defined in fpnew_pkg.sv?

Thanks in advance for your help!

MikeOpenHWGroup, 2025-04-21:
Hi @omarsalaheldinnn, thanks for your interest in the CV32E40P.  No, line 91 is correct as is: `PipeConfig: default: fpnew_pkg::AFTER }`.  Can you provide:
- the _exact_ set of commands you invoked to produce this error, starting with `git clone...`
- the version of Synopsys DC you are using?

omarsalaheldinnn, 2025-04-24:
Hi @MikeOpenHWGroup, thanks for the clarification!

1. I'm working on PULPissimo, which uses a tool called Bender to handle dependencies and setup the environment.

2. I'm using DC 2018

MikeOpenHWGroup, 2025-04-24:
> I'm working on PULPissimo, which uses a tool called Bender to handle dependencies and setup the environment.

Sorry, I cannot help you there.  If you know someone at UniBo and/or ETHZ perhaps they can help.

> I'm using DC 2018

That is a very old version, and the cv32e40p tends to use newer code constructs that are not well supported by older tools.  If you have access to a newer version of DC it would be a good idea to try that.

omarsalaheldinnn, 2025-04-25:
Thank you for your help and the clarification