openhwfoundation/cv32e40p #1044: Constraint Application Error: Ports Not Found During CV32E40P Synthesis

AlixYehia, 2025-05-05:
During the synthesis of the CV32E40P core, we encountered an error stating that some ports targeted by the constraints are not found in the design. Could this be due to a mismatch between the constraint file and the current RTL, such as outdated or incorrect port names?

[image]

MikeOpenHWGroup, 2025-05-06:
Hi @AlixYehia, thanks for your interest in CV32E40P.  Can you provide the _exact_ command lines you used to obtain these results, starting with `git clone ...`.  Also, have you read the [synthesis guidelines](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/integration.html#synthesis-guidelines) in the User Manual?

AlixYehia, 2025-05-06:
Yes, I have read the synthesis guidelines in the User Manual.
The issue appears to occur in the following section of the **[cv32e40p_core.sdc](https://github.com/openhwgroup/cv32e40p/blob/master/constraints/cv32e40p_core.sdc)** file:

```
# IRQ Output ports
set irq_output_ports [list \
    irq_ack_o \
    irq_id_o* \
]

# Early Input ports (ideally from register)
set early_input_ports [list \
    debug_req_i \
    boot_addr_i* \
    mtvec_addr_i* \
    dm_halt_addr_i* \
    hart_id_i* \
    dm_exception_addr_i* \
]

# RISC-V OBI Input ports
set obi_input_ports [list \
    instr_gnt_i \
    instr_rvalid_i \
    instr_rdata_i* \
    data_gnt_i \
    data_rvalid_i \
    data_rdata_i* \
]

# RISC-V OBI Output ports
set obi_output_ports [list \
    instr_req_o \
    instr_addr_o* \
    data_req_o \
    data_we_o \
    data_be_o* \
    data_addr_o* \
    data_wdata_o* \
]

########### Defining Default I/O constraints ###################

set all_other_input_ports  [remove_from_collection [all_inputs]  [get_ports [list $all_clock_ports $obi_input_ports $irq_input_ports $early_input_ports]]]

set all_other_output_ports [remove_from_collection [all_outputs] [get_ports [list $all_clock_ports $obi_output_ports $sleep_output_ports $irq_output_ports]]]

```

Here are the files I used for synthesis:

```

analyze -format $file_format cv32e40p_pkg.sv
analyze -format $file_format cv32e40p_fpu_pkg.sv
analyze -format $file_format cv32e40p_apu_core_pkg.sv

analyze -format $file_format fpnew_pkg.sv

analyze -format $file_format cv32e40p_aligner.sv
analyze -format $file_format cv32e40p_alu.sv
analyze -format $file_format cv32e40p_alu_div.sv
analyze -format $file_format cv32e40p_apu_disp.sv
analyze -format $file_format cv32e40p_compressed_decoder.sv
analyze -format $file_format cv32e40p_controller.sv
analyze -format $file_format cv32e40p_core.sv
analyze -format $file_format cv32e40p_cs_registers.sv
analyze -format $file_format cv32e40p_decoder.sv
analyze -format $file_format cv32e40p_ex_stage.sv
analyze -format $file_format cv32e40p_ff_one.sv
analyze -format $file_format cv32e40p_fifo.sv
analyze -format $file_format cv32e40p_fp_wrapper.sv
analyze -format $file_format cv32e40p_hwloop_regs.sv
analyze -format $file_format cv32e40p_id_stage.sv
analyze -format $file_format cv32e40p_if_stage.sv
analyze -format $file_format cv32e40p_int_controller.sv
analyze -format $file_format cv32e40p_load_store_unit.sv
analyze -format $file_format cv32e40p_mult.sv
analyze -format $file_format cv32e40p_obi_interface.sv
analyze -format $file_format cv32e40p_popcnt.sv
analyze -format $file_format cv32e40p_prefetch_buffer.sv
analyze -format $file_format cv32e40p_prefetch_controller.sv
analyze -format $file_format cv32e40p_register_file_ff.sv
analyze -format $file_format cv32e40p_clock_gate.sv
analyze -format $file_format cv32e40p_sleep_unit.sv

analyze -format $file_format cv32e40p_top.sv

elaborate -lib work cv32e40p_top

```

MikeOpenHWGroup, 2025-05-07:
Ah, now I get it.  IIRC, the sdc file is simply a constraints file.  It does not include all the commands to ready in the files, etc.  Sorry for the confusion.  If you'd like to discuss why we do things that way, reach out to me on [MatterMost](https://mattermost.openhwgroup.org/), the OpenHW chat tool.

closed 2025-05-07 by AlixYehia